FujiHeatPump/esphome-fujitsu #2: Issues uploading to ESP32 device.

sag007, 2022-07-15:
Hey All,

I am new to HA let alone esphome and have been having issues trying to get this onto an esp32 device. If anyone could help a newbe
I would be eternally grateful. Ever since installing this $5k ducted ac unit I have been annoyed at the fact that if I had purchased a cheaper unit, it probably would have come with wifi for me to be able to connect it to my network and start controlling it. You think buying a good brand you would get the good stuff but unfortunately this isn't always the case. I looked at the options of Anywair however as you would expect the $330 unit doesn't work with my unit however as you would expect the $1600 package does.

The process I have followed thus far was in the latest version of both HA and the ESPHome integration was to open the ESPHome web gui, adopt the esp32 device, then copied the files into the directory as mention and install however it always fails to compile and upload to the device.

If anyone oculd offer any suggestions it would be greatly appreciated.

Regards,

Nathan

sag007, 2022-07-18:
Hey Guys,

So I finally seemed to have uploaded this to the ESP32 device but checking the debug on the usb serial I get the following;

[0;32m[I][logger:243]: Log initialized[0m
[0;35m[C][ota:461]: There have been 6 suspected unsuccessful boot attempts.[0m
[0;36m[D][esp32.preferences:114]: Saving preferences to flash...[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;36m[D][fuji:028]: Fuji initialized[0m
[0;36m[D][fuji:035]: starting task[0m
[0;36m[D][fuji:010]: reached task[0m
[0;36m[D][fuji:011]: serialTask started on core 1[0m

I haven't made the circuit nor connected the unit to the unit but not sure why the unit keeps rebooting.
Also once the  unit boots should there be entities that then show up in HA? I dont see anything at all.

Regards,

Nathan

MaoZ1993, 2022-08-01:
Hi there

I think I encountered the same issue, after flashing into my esp32-wroom-32, it was trapped into a booting loop, and never brought back on.

Boot loop:
[image]

warning while flashing
[image]

closed 2022-09-13 by rabbit-aaron

tevaadarelsol, 2022-12-20:
I am also facing this issue. Any way to solve it, please?

Thank you!

tevaadarelsol, 2023-01-15:
Hello @sag007 and @MaoZ1993, did you managed to get it working? I'm facing the same issue. Thank you!