FujiHeatPump/esphome-fujitsu #4: Issues uploading to ESP32 device

tevaadarelsol, 2022-12-20:
Hello mate, 

Trying to upload your code to an ESP32-WROOM-32 device causes bootloop. Logs attached. Is there any way to solve it?

Thank you so much, I really appreciate your work as you bring us the possibility to make our aircons intelligen without having to waste so much money by replacing them!

E (10332) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10332) task_wdt:  - loopTask (CPU 1)
E (10332) task_wdt: Tasks currently running:
E (10332) task_wdt: CPU 0: IDLE
E (10332) task_wdt: CPU 1: FujiTask
E (10332) task_wdt: Aborting.

abort() was called at PC 0x400f5460 on core 0

Backtrace:0x400836dd:0x3ffbe9bc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][fuji:028]: Fuji initialized
[D][fuji:035]: starting task
[D][fuji:010]: reached task
[D][fuji:011]: serialTask started on core 1
E (10344) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10344) task_wdt:  - loopTask (CPU 1)
E (10344) task_wdt: Tasks currently running:
E (10344) task_wdt: CPU 0: IDLE
E (10344) task_wdt: CPU 1: FujiTask
E (10344) task_wdt: Aborting.

abort() was called at PC 0x400f5460 on core 0

Backtrace:0x400836dd:0x3ffbe9bc |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 2 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][fuji:028]: Fuji initialized
[D][fuji:035]: starting task
[... 294 lines cut ...]
[I][app:062]: setup() finished successfully!
[I][ota:489]: Waiting for OTA attempt.
[I][app:102]: ESPHome version 2022.12.3 compiled on Dec 20 2022, 18:52:42
[C][wifi:504]: WiFi:
[C][wifi:362]:   Local MAC: 78:21:84:BB:2D:E4
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.20.60
[C][wifi:366]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'fujitsu'
[C][wifi:369]:   Signal strength: -77 dB ▂▄▆█
[C][wifi:373]:   Channel: 1
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.20.1
[C][wifi:376]:   DNS1: 0.0.0.0
[C][wifi:377]:   DNS2: 0.0.0.0
[C][logger:293]: Logger:
[C][logger:294]:   Level: DEBUG
[C][logger:295]:   Log Baud Rate: 115200
[C][logger:296]:   Hardware UART: UART0
[C][captive_portal:088]: Captive Portal:
[C][mdns:103]: mDNS:
[C][mdns:104]:   Hostname: fujitsu
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.20.60:3232
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts

thaenan, 2023-01-14:
Hi @tevaadarelsol, did you manage to get it running? I just got my ESP2-WROOM-32E today and I have exactly the same problem as yours.
Thanks!

tevaadarelsol, 2023-01-14:
Hello @thaenan. I had no time to put my hands on It again, but was trying to get It working without success.... It would be nice if @rabbit-aaron could give us a helping hand.

rabbit-aaron, 2023-01-24:
Sorry, I'm not too familiar with the ESP32 variants, I used ESP32-S, if you could get your hands on one that'll probably work.

I am no expert on the ESPHome either so there's little I could do. Would be nice if someone figure this out and could post an answer here.

thaenan, 2023-01-24:
Hi, actually this solution proposed in the HA community forum worked for me: https://community.home-assistant.io/t/fujitsu-ac-heat-pump-integration-via-esphome-esp32/407610/102?u=dosreism.

`modified in the library file “FujitsuClimate.cpp” in the setup function the
xTaskCreatePinnedToCore() statament changing only the priority parameter to 1`

tevaadarelsol, 2023-01-26:
> Hi, actually this solution proposed in the HA community forum worked for me: https://community.home-assistant.io/t/fujitsu-ac-heat-pump-integration-via-esphome-esp32/407610/102?u=dosreism. [...]

THANK YOU SO MUCH @thaenan !! This was the problem, I think I've searched on ALL the internet looking for solutions except for the most obvious place: the own development thread (facepalm).

After changing this parameter I was able to upload it. However the ESP32 didn't communicate with the aircon. At the same thread someone explained there is a dip switch that must be changed and, after that, I managed to get It working flawlessly.

Huge thanks to @rabbit-aaron for the contribution and @thaenan for pointing me on the right direction!

closed 2023-01-26 by tevaadarelsol

rabbit-aaron, 2023-02-07:
Another reason why you might have bootloop is due to the LIN transceiver not being correctly connected, note the Rx connects to Rx and Tx connects to Tx, unlike other UART connections.