FujiHeatPump/esphome-fujitsu #5: Add fan mode Quiet

fpitre75, 2023-02-07:
It is possible to add fan mode Quiet in this project or new project for those with these support mode, mine support Fan quiet mode with wall remote and IR remote but not with esphome-fujitsu project.

In file (FujiHeatPump.h) we see some reference for mode fan quiet ?

enum class FujiFanMode : byte {
    FAN_AUTO = 0,
    FAN_QUIET = 1,
    FAN_LOW = 2,
    FAN_MEDIUM = 3,
    FAN_HIGH = 4

But in file (FujitsuClimate.cpp) no reference

optional<climate::ClimateFanMode> FujitsuClimate::fujiToEspFanMode(
    FujiFanMode fujiFanMode) {
    if (fujiFanMode == FujiFanMode::FAN_AUTO) {
        return climate::ClimateFanMode::CLIMATE_FAN_AUTO;
    }

    if (fujiFanMode == FujiFanMode::FAN_HIGH) {
        return climate::ClimateFanMode::CLIMATE_FAN_HIGH;
    }

    if (fujiFanMode == FujiFanMode::FAN_MEDIUM) {
        return climate::ClimateFanMode::CLIMATE_FAN_MEDIUM;
    }

    if (fujiFanMode == FujiFanMode::FAN_LOW) {
        return climate::ClimateFanMode::CLIMATE_FAN_LOW;
    }

    return {};

I don't have skill for coding that but that will be appreciate add this mode

tevaadarelsol, 2023-03-16:
My pump does nota support quiet mode so I cannot test It, but try to add this just before the return{};

if (fujiFanMode == FujiFanMode::FAN_QUIET) {
    return climate::ClimateFanMode::CLIMATE_FAN_QUIET;
}

Maybe It works

fpitre75, 2023-03-16:
Thanks it works , I try before but error when compiling , simple syntax error
I  add your code in FujitsuClimate.cpp

if (fujiFanMode == FujiFanMode::FAN_QUIET) {
return climate::ClimateFanMode::CLIMATE_FAN_QUIET;
}

This line to

if (espFanMode == climate::ClimateFanMode::CLIMATE_FAN_QUIET) {
        return FujiFanMode::FAN_QUIET;
    }

And 

  traits.set_supported_fan_modes(
        {climate::CLIMATE_FAN_AUTO, climate::CLIMATE_FAN_LOW,**_climate::CLIMATE_FAN_QUIET,_**
         climate::CLIMATE_FAN_MEDIUM, climate::CLIMATE_FAN_HIGH});

Little detail in Home Assistant Quiet mode first letter is write in small caps ? but no big deal, it works  SUPER

[image]

Thanks you so much.

closed 2023-03-16 by fpitre75

uspino2, 2023-11-06:
Thank you for this. Did you solve the small caps issue? And is there a way to get "Quiet" mode between Auto and Low, instead of after "High"?

fpitre75, 2023-11-06:
Quiet mode  works by adding code in FujitsuClimate.cpp 

if (fujiFanMode == FujiFanMode::FAN_QUIET) {
return climate::ClimateFanMode::CLIMATE_FAN_QUIET;
}

For small caps issue that was resolve  probably  in HA update.

uspino2, 2023-11-06:
Thank you, I did that and it works. What I was wondering is how to show the correct order in the drop-down list: quiet-low-medium-high, instead of how it shows now (low-medium-high-quiet). It's just a cosmetic thing...

fpitre75, 2023-11-06:
For  Fan Speed menu look like sort by alphabet in HA , and for Quiet speed no icon probably because HA doesn't have icon for this fan speed for generic thermostat. But like you said only cosmetic, hope the code will update in time for issue that sometime command is not send correctly.

[image]