FujiHeatPump/esphome-fujitsu #10: testing esp-idf-port on Wemos C3 mini

patvdleer, 2023-05-19:
@dgrnbrg

On the Wemos C3 mini there is no UART2/Serial2, is it really required?

```
In file included from src/esphome/components/fujitsu_heat_pump/FujitsuClimate.cpp:1:
src/esphome/components/fujitsu_heat_pump/FujitsuClimate.h: In member function 'void esphome::fujitsu::FujitsuClimate::late_start()':
src/esphome/components/fujitsu_heat_pump/FujitsuClimate.h:24:32: error: 'UART_NUM_2' was not declared in this scope
         this->heatPump.connect(UART_NUM_2, !this->is_master_, rx, tx);
                                ^~~~~~~~~~
src/esphome/components/fujitsu_heat_pump/FujitsuClimate.h:24:32: note: suggested alternative: 'UART_NUM_0'
         this->heatPump.connect(UART_NUM_2, !this->is_master_, rx, tx);
                                ^~~~~~~~~~
                                UART_NUM_0
```

ref: https://github.com/FujiHeatPump/esphome-fujitsu/blob/cb513df53a18c4120378a1a86ee49a0a11728a08/components/fujitsu_heat_pump/FujitsuClimate.h#LL24C32-L24C42

ESPHOME: 2023.5.1

```yaml
esphome:
  name: airco_bedroom
  friendly_name: airco_bedroom
  
esp32:
  board: lolin_c3_mini
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

web_server:
  port: 80
  include_internal: True

# Add the dependency to pull in the fujitsu heat pump component
external_components:
  - source: github://FujiHeatPump/esphome-fujitsu@esp-idf-port

climate:
- platform: fujitsu_heat_pump
  name: Bedroom airco
  # The rest of these options are optional
  tx_pin: 4
  rx_pin: 5
  remote_temperature_id: some_temp_sensor # TODO see FujitsuClimate.cpp to correct the implementation
  master: false

# Pull a sensor in from homeassistant
sensor:
  - platform: homeassistant
    entity_id: sensor.nope_nope_nope
    id: some_temp_sensor
```

dgrnbrg, 2023-05-19:
It isn’t required, but also, I haven’t ported the esphome uart stuff, so there’s nothing to protect you from binding a uart to 2 things :)

feel free to change it and see if that works. Does that model have 2 cores?

dgrnbrg, 2023-05-19:
Also, you should not use the remote temperature sensor functionality—that isn’t implemented correctly yet (reverse engineering is ongoing).

patvdleer, 2023-05-20:
No it's single core, C3 is single core, the S3 is dual core.

https://www.wemos.cc/en/latest/c3/c3_mini.html
https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf

https://www.wemos.cc/en/latest/s3/s3_mini.html
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf