FujiHeatPump/esphome-fujitsu #17: fan-only mode changes the target temperature to 0°C

rehakf, 2024-06-09:
When I set climate:fan-only mode in homeassistant it changes the target temperature to 0°C
(even if target min=16°C and max=30°C).
Later when I set climate:cool mode, the target temperature is set to 0°C, then in a second it raises to 18°C, but the last used cooling target temp is lost (eg. 24°C).

When using remote control this doesn’t happen.
Wired remote doesn’t make any changes in the target temperature when using fan mode (it just doesn’t display the target temp).
Later when I set climate:cool mode, the last used cooling target temperature remains (eg. 24°C).

rehakf, 2024-06-12:
I found that this is probably due to the wired controller. 
When set to fan_only mode (no matter if by HA or wired ctrl) then the controller set the target temp to 0.
Later if set to cool mode things are different:
- if set to cool by HA - cooling start with target temp=0°C immediately changed to 18°C
- if set to cool by wired ctrl - cooling start with the last known target temperature used for cooling which is probably memorised in the controller

rehakf, 2024-06-12:
Log fron esphome with my comments:
`[17:51:51][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 11 2024, 16:33:12
[17:51:51][C][wifi:580]: WiFi:
[17:51:52][C][wifi:408]:   Local MAC: xxxxx
[17:51:52][C][wifi:413]:   SSID: [redacted]
[17:51:52][C][wifi:416]:   IP Address: 10.0.0.9
[17:51:52][C][wifi:420]:   BSSID: [redacted]
[17:51:52][C][wifi:421]:   Hostname: 'fujitsu'
[17:51:52][C][wifi:423]:   Signal strength: -63 dB ▂▄▆█
[17:51:52][C][wifi:427]:   Channel: 10
[17:51:52][C][wifi:428]:   Subnet: 255.255.255.0
[17:51:52][C][wifi:429]:   Gateway: 10.0.0.138
[17:51:52][C][wifi:430]:   DNS1: 10.0.0.138
[17:51:52][C][wifi:431]:   DNS2: 0.0.0.0
[17:51:52][C][logger:185]: Logger:
[17:51:52][C][logger:186]:   Level: DEBUG
[17:51:52][C][logger:188]:   Log Baud Rate: 115200
[17:51:52][C][logger:189]:   Hardware UART: UART0
[17:51:52][C][captive_portal:088]: Captive Portal:
[17:51:52][C][web_server:173]: Web Server:
[17:51:52][C][web_server:174]:   Address: fujitsu.local:80
[17:51:52][C][mdns:115]: mDNS:
[17:51:52][C][mdns:116]:   Hostname: fujitsu
[17:51:52][C][ota:096]: Over-The-Air Updates:
[17:51:52][C][ota:097]:   Address: fujitsu.local:3232
[17:51:53][C][ota:103]:   OTA version: 2.
[17:51:53][C][api:144]:   Using noise encryption: NO
[17:51:54][D][api:102]: Accepted 10.0.0.8
[17:51:54][D][api.connection:1321]: Home Assistant 2024.6.1 (10.0.0.8): Connected successfully
[17:52:14][D][climate:011]: 'Fujitsu' - Setting
[17:52:14][D][climate:040]:   Target Temperature: 24.00
[17:52:14][D][fuji:234]: Fuji setting temperature 24.000000
[17:52:15][D][fuji:145]: ctrl temp 24 vs my temp 4194303
[17:52:15][D][fuji:201]: publishing state
---start---
[17:52:15][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:15][D][climate:399]:   Mode: FAN_ONLY
[17:52:15][D][climate:404]:   Fan Mode: AUTO
[17:52:15][D][climate:419]:   Current Temperature: 27.00°C
[17:52:15][D][climate:425]:   Target Temperature: 24.00°C
[17:52:16][D][fuji:145]: ctrl temp 0 vs my temp 4194303 
[17:52:16][D][fuji:201]: publishing state
[17:52:16][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:16][D][climate:399]:   Mode: FAN_ONLY
[17:52:16][D][climate:404]:   Fan Mode: AUTO
[17:52:16][D][climate:419]:   Current Temperature: 27.00°C
[17:52:16][D][climate:425]:   Target Temperature: 0.00°C
---HA set to cool---
[17:55:27][D][climate:011]: 'Fujitsu' - Setting
[17:55:27][D][climate:015]:   Mode: COOL
[17:55:27][D][fuji:213]: Fuji setting mode 2
[17:55:28][D][fuji:155]: ctrl mode 2 vs my mode 4
[17:55:28][D][fuji:201]: publishing state
[17:55:28][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:28][D][climate:399]:   Mode: COOL
[17:55:28][D][climate:404]:   Fan Mode: AUTO
[17:55:28][D][climate:419]:   Current Temperature: 27.00°C
[17:55:28][D][climate:425]:   Target Temperature: 0.00°C
[17:55:29][D][fuji:145]: ctrl temp 18 vs my temp 4194303
[17:55:29][D][fuji:201]: publishing state
[17:55:29][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:29][D][climate:399]:   Mode: COOL
[17:55:29][D][climate:404]:   Fan Mode: AUTO
[17:55:29][D][climate:419]:   Current Temperature: 27.00°C
[17:55:29][D][climate:425]:   Target Temperature: 18.00°C
--HA set target temp to 22---
[17:55:31][D][climate:011]: 'Fujitsu' - Setting
[17:55:31][D][climate:040]:   Target Temperature: 22.00
[17:55:31][D][fuji:234]: Fuji setting temperature 22.000000
[17:55:32][D][fuji:145]: ctrl temp 22 vs my temp 4194303
[... 36 lines cut ...]
[18:06:28][D][fuji:155]: ctrl mode 2 vs my mode 0
[18:06:28][D][fuji:201]: publishing state
[18:06:28][D][climate:396]: 'Fujitsu' - Sending state:
[18:06:28][D][climate:399]:   Mode: COOL
[18:06:28][D][climate:404]:   Fan Mode: AUTO
[18:06:28][D][climate:419]:   Current Temperature: 27.00°C
[18:06:28][D][climate:425]:   Target Temperature: 22.00°C
--wired controller set to fan only--
[18:08:57][D][fuji:145]: ctrl temp 0 vs my temp 4194303
[18:08:57][D][fuji:155]: ctrl mode 4 vs my mode 2
[18:08:57][D][fuji:201]: publishing state
[18:08:57][D][climate:396]: 'Fujitsu' - Sending state:
[18:08:57][D][climate:399]:   Mode: FAN_ONLY
[18:08:57][D][climate:404]:   Fan Mode: AUTO
[18:08:57][D][climate:419]:   Current Temperature: 27.00°C
[18:08:57][D][climate:425]:   Target Temperature: 0.00°C
--wired controller set to cool--
[18:10:30][D][fuji:145]: ctrl temp 22 vs my temp 4194303
[18:10:30][D][fuji:155]: ctrl mode 2 vs my mode 4
[18:10:30][D][fuji:201]: publishing state
[18:10:30][D][climate:396]: 'Fujitsu' - Sending state:
[18:10:30][D][climate:399]:   Mode: COOL
[18:10:30][D][climate:404]:   Fan Mode: AUTO
[18:10:30][D][climate:419]:   Current Temperature: 27.00°C
[18:10:30][D][climate:425]:   Target Temperature: 22.00°C`

rehakf, 2024-06-12:
Log from esphome:

`[17:51:51][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 11 2024, 16:33:12
[17:51:51][C][wifi:580]: WiFi:
[17:51:52][C][wifi:408]:   Local MAC: xxxxx
[17:51:52][C][wifi:413]:   SSID: [redacted]
[17:51:52][C][wifi:416]:   IP Address: 10.0.0.9
[17:51:52][C][wifi:420]:   BSSID: [redacted]
[17:51:52][C][wifi:421]:   Hostname: 'fujitsu'
[17:51:52][C][wifi:423]:   Signal strength: -63 dB ▂▄▆█
[17:51:52][C][wifi:427]:   Channel: 10
[17:51:52][C][wifi:428]:   Subnet: 255.255.255.0
[17:51:52][C][wifi:429]:   Gateway: 10.0.0.138
[17:51:52][C][wifi:430]:   DNS1: 10.0.0.138
[17:51:52][C][wifi:431]:   DNS2: 0.0.0.0
[17:51:52][C][logger:185]: Logger:
[17:51:52][C][logger:186]:   Level: DEBUG
[17:51:52][C][logger:188]:   Log Baud Rate: 115200
[17:51:52][C][logger:189]:   Hardware UART: UART0
[17:51:52][C][captive_portal:088]: Captive Portal:
[17:51:52][C][web_server:173]: Web Server:
[17:51:52][C][web_server:174]:   Address: fujitsu.local:80
[17:51:52][C][mdns:115]: mDNS:
[17:51:52][C][mdns:116]:   Hostname: fujitsu
[17:51:52][C][ota:096]: Over-The-Air Updates:
[17:51:52][C][ota:097]:   Address: fujitsu.local:3232
[17:51:53][C][ota:103]:   OTA version: 2.
[17:51:53][C][api:144]:   Using noise encryption: NO
[17:51:54][D][api:102]: Accepted 10.0.0.8
[17:51:54][D][api.connection:1321]: Home Assistant 2024.6.1 (10.0.0.8): Connected successfully
[17:52:14][D][climate:011]: 'Fujitsu' - Setting
[17:52:14][D][climate:040]:   Target Temperature: 24.00
[17:52:14][D][fuji:234]: Fuji setting temperature 24.000000
[17:52:15][D][fuji:145]: ctrl temp 24 vs my temp 4194303
[17:52:15][D][fuji:201]: publishing state
---start---
[17:52:15][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:15][D][climate:399]:   Mode: FAN_ONLY
[17:52:15][D][climate:404]:   Fan Mode: AUTO
[17:52:15][D][climate:419]:   Current Temperature: 27.00°C
[17:52:15][D][climate:425]:   Target Temperature: 24.00°C
[17:52:16][D][fuji:145]: ctrl temp 0 vs my temp 4194303
[17:52:16][D][fuji:201]: publishing state
[17:52:16][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:16][D][climate:399]:   Mode: FAN_ONLY
[17:52:16][D][climate:404]:   Fan Mode: AUTO
[17:52:16][D][climate:419]:   Current Temperature: 27.00°C
[17:52:16][D][climate:425]:   Target Temperature: 0.00°C
---HA set to cool---
[17:55:27][D][climate:011]: 'Fujitsu' - Setting
[17:55:27][D][climate:015]:   Mode: COOL
[17:55:27][D][fuji:213]: Fuji setting mode 2
[17:55:28][D][fuji:155]: ctrl mode 2 vs my mode 4
[17:55:28][D][fuji:201]: publishing state
[17:55:28][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:28][D][climate:399]:   Mode: COOL
[17:55:28][D][climate:404]:   Fan Mode: AUTO
[17:55:28][D][climate:419]:   Current Temperature: 27.00°C
[17:55:28][D][climate:425]:   Target Temperature: 0.00°C
[17:55:29][D][fuji:145]: ctrl temp 18 vs my temp 4194303
[17:55:29][D][fuji:201]: publishing state
[17:55:29][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:29][D][climate:399]:   Mode: COOL
[17:55:29][D][climate:404]:   Fan Mode: AUTO
[17:55:29][D][climate:419]:   Current Temperature: 27.00°C
[17:55:29][D][climate:425]:   Target Temperature: 18.00°C
--HA set target t to 22---
[17:55:31][D][climate:011]: 'Fujitsu' - Setting
[17:55:31][D][climate:040]:   Target Temperature: 22.00
[17:55:31][D][fuji:234]: Fuji setting temperature 22.000000
[... 35 lines cut ...]
[18:06:28][D][fuji:155]: ctrl mode 2 vs my mode 0
[18:06:28][D][fuji:201]: publishing state
[18:06:28][D][climate:396]: 'Fujitsu' - Sending state:
[18:06:28][D][climate:399]:   Mode: COOL
[18:06:28][D][climate:404]:   Fan Mode: AUTO
[18:06:28][D][climate:419]:   Current Temperature: 27.00°C
[18:06:28][D][climate:425]:   Target Temperature: 22.00°C
--wired controller set to fan only--
[18:08:57][D][fuji:145]: ctrl temp 0 vs my temp 4194303
[18:08:57][D][fuji:155]: ctrl mode 4 vs my mode 2
[18:08:57][D][fuji:201]: publishing state
[18:08:57][D][climate:396]: 'Fujitsu' - Sending state:
[18:08:57][D][climate:399]:   Mode: FAN_ONLY
[18:08:57][D][climate:404]:   Fan Mode: AUTO
[18:08:57][D][climate:419]:   Current Temperature: 27.00°C
[18:08:57][D][climate:425]:   Target Temperature: 0.00°C
--**wired controller** set to cool--
[18:10:30][D][fuji:145]: ctrl temp 22 vs my temp 4194303
[18:10:30][D][fuji:155]: ctrl mode 2 vs my mode 4
[18:10:30][D][fuji:201]: publishing state
[18:10:30][D][climate:396]: 'Fujitsu' - Sending state:
[18:10:30][D][climate:399]:   Mode: COOL
[18:10:30][D][climate:404]:   Fan Mode: AUTO
[18:10:30][D][climate:419]:   Current Temperature: 27.00°C
[18:10:30][D][climate:425]:   Target Temperature: 22.00°C`

rehakf, 2024-06-12:
Log from esphome:

`[17:51:51][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 11 2024, 16:33:12
[17:51:51][C][wifi:580]: WiFi:
[17:51:52][C][wifi:408]:   Local MAC: xxxxx
[17:51:52][C][wifi:413]:   SSID: [redacted]
[17:51:52][C][wifi:416]:   IP Address: 10.0.0.9
[17:51:52][C][wifi:420]:   BSSID: [redacted]
[17:51:52][C][wifi:421]:   Hostname: 'fujitsu'
[17:51:52][C][wifi:423]:   Signal strength: -63 dB ▂▄▆█
[17:51:52][C][wifi:427]:   Channel: 10
[17:51:52][C][wifi:428]:   Subnet: 255.255.255.0
[17:51:52][C][wifi:429]:   Gateway: 10.0.0.138
[17:51:52][C][wifi:430]:   DNS1: 10.0.0.138
[17:51:52][C][wifi:431]:   DNS2: 0.0.0.0
[17:51:52][C][logger:185]: Logger:
[17:51:52][C][logger:186]:   Level: DEBUG
[17:51:52][C][logger:188]:   Log Baud Rate: 115200
[17:51:52][C][logger:189]:   Hardware UART: UART0
[17:51:52][C][captive_portal:088]: Captive Portal:
[17:51:52][C][web_server:173]: Web Server:
[17:51:52][C][web_server:174]:   Address: fujitsu.local:80
[17:51:52][C][mdns:115]: mDNS:
[17:51:52][C][mdns:116]:   Hostname: fujitsu
[17:51:52][C][ota:096]: Over-The-Air Updates:
[17:51:52][C][ota:097]:   Address: fujitsu.local:3232
[17:51:53][C][ota:103]:   OTA version: 2.
[17:51:53][C][api:144]:   Using noise encryption: NO
[17:51:54][D][api:102]: Accepted 10.0.0.8
[17:51:54][D][api.connection:1321]: Home Assistant 2024.6.1 (10.0.0.8): Connected successfully
[17:52:14][D][climate:011]: 'Fujitsu' - Setting
[17:52:14][D][climate:040]:   Target Temperature: 24.00
[17:52:14][D][fuji:234]: Fuji setting temperature 24.000000
[17:52:15][D][fuji:145]: ctrl temp 24 vs my temp 4194303
[17:52:15][D][fuji:201]: publishing state
---start---
[17:52:15][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:15][D][climate:399]:   Mode: FAN_ONLY
[17:52:15][D][climate:404]:   Fan Mode: AUTO
[17:52:15][D][climate:419]:   Current Temperature: 27.00°C
[17:52:15][D][climate:425]:   Target Temperature: 24.00°C
[17:52:16][D][fuji:145]: ctrl temp 0 vs my temp 4194303
[17:52:16][D][fuji:201]: publishing state
[17:52:16][D][climate:396]: 'Fujitsu' - Sending state:
[17:52:16][D][climate:399]:   Mode: FAN_ONLY
[17:52:16][D][climate:404]:   Fan Mode: AUTO
[17:52:16][D][climate:419]:   Current Temperature: 27.00°C
[17:52:16][D][climate:425]:   Target Temperature: 0.00°C
---HA set to cool---
[17:55:27][D][climate:011]: 'Fujitsu' - Setting
[17:55:27][D][climate:015]:   Mode: COOL
[17:55:27][D][fuji:213]: Fuji setting mode 2
[17:55:28][D][fuji:155]: ctrl mode 2 vs my mode 4
[17:55:28][D][fuji:201]: publishing state
[17:55:28][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:28][D][climate:399]:   Mode: COOL
[17:55:28][D][climate:404]:   Fan Mode: AUTO
[17:55:28][D][climate:419]:   Current Temperature: 27.00°C
[17:55:28][D][climate:425]:   Target Temperature: 0.00°C
[17:55:29][D][fuji:145]: ctrl temp 18 vs my temp 4194303
[17:55:29][D][fuji:201]: publishing state
[17:55:29][D][climate:396]: 'Fujitsu' - Sending state:
[17:55:29][D][climate:399]:   Mode: COOL
[17:55:29][D][climate:404]:   Fan Mode: AUTO
[17:55:29][D][climate:419]:   Current Temperature: 27.00°C
[17:55:29][D][climate:425]:   Target Temperature: 18.00°C
--HA set target t to 22---
[17:55:31][D][climate:011]: 'Fujitsu' - Setting
[17:55:31][D][climate:040]:   Target Temperature: 22.00
[17:55:31][D][fuji:234]: Fuji setting temperature 22.000000
[... 36 lines cut ...]
[18:06:28][D][fuji:155]: ctrl mode 2 vs my mode 0
[18:06:28][D][fuji:201]: publishing state
[18:06:28][D][climate:396]: 'Fujitsu' - Sending state:
[18:06:28][D][climate:399]:   Mode: COOL
[18:06:28][D][climate:404]:   Fan Mode: AUTO
[18:06:28][D][climate:419]:   Current Temperature: 27.00°C
[18:06:28][D][climate:425]:   Target Temperature: 22.00°C
--wired controller set to fan only--
[18:08:57][D][fuji:145]: ctrl temp 0 vs my temp 4194303
[18:08:57][D][fuji:155]: ctrl mode 4 vs my mode 2
[18:08:57][D][fuji:201]: publishing state
[18:08:57][D][climate:396]: 'Fujitsu' - Sending state:
[18:08:57][D][climate:399]:   Mode: FAN_ONLY
[18:08:57][D][climate:404]:   Fan Mode: AUTO
[18:08:57][D][climate:419]:   Current Temperature: 27.00°C
[18:08:57][D][climate:425]:   Target Temperature: 0.00°C
--wired controller set to cool--
[18:10:30][D][fuji:145]: ctrl temp 22 vs my temp 4194303
[18:10:30][D][fuji:155]: ctrl mode 2 vs my mode 4
[18:10:30][D][fuji:201]: publishing state
[18:10:30][D][climate:396]: 'Fujitsu' - Sending state:
[18:10:30][D][climate:399]:   Mode: COOL
[18:10:30][D][climate:404]:   Fan Mode: AUTO
[18:10:30][D][climate:419]:   Current Temperature: 27.00°C
[18:10:30][D][climate:425]:   Target Temperature: 22.00°C`

rabbit-aaron, 2024-09-14:
i cannot replicate this issue so can't help you out here.

closed 2024-09-14 by rabbit-aaron